Zepmanbc/creopyson #98: `display_data` parameter format missing for `drawing_create_gen_view` method

babyhyf, 2025-06-27:
When using the `drawing_create_gen_view` method, I'm unclear about the specific format and available options for the `display_data` parameter. The current documentation only states that this parameter is a dictionary, but doesn't list supported key-value pairs.

Here's a relevant code snippet:
```python
display_data = {
    # Unsure about which parameters to pass here
    "hidden_line": "hidden",  # Example parameter, unsure if correct
    "view_scale": 1.0         # Example parameter, unsure if correct
}

c.drawing_create_gen_view(
    model_view="FRONT",
    point={"x": 100, "y": 100},
    display_data=display_data
)
```

**Questions:**  
1. What specific parameters does `display_data` support?
2. What are the value ranges and meanings for each parameter?
3. Are there any parameter dependencies or version compatibility issues?

**Environment Information:**  
- creopyson version: latest
- Creo version:8.0
- Python version: 3.11.9

Thank you for your assistance!

babyhyf, 2025-06-27:
@Zepmanbc

Zepmanbc, 2025-07-01:
You can choose if the scale is shown or not in the drowing properties, this is a Creo affair, not Creoson/Creopyson

adama2000, 2025-07-06:
To answer your earlier question, "What specific parameters does display_data support?", if you go to the link @Zepmanbc  gave (http://localhost:9056/functions.html) and use the dropdown to view "object : ViewDisplayData" that will give you the details of what fields are available and what their valid values are.