Zepmanbc/creopyson #99: Connection reset error debug

bperryrm, 2025-08-27:
I'm running a lot of calls into creopyson and I'm getting a random connection error on long operations with many calls. The error is below and I'm assuming I'm just overloading the server and it's kicking me out because of that. 
---
Error: ('Connection aborted.', ConnectionResetError(10054, 'An existing connection was forcibly closed by the remote host', None, 10054, None))
---
Anyone have some guidance on how to either recover the connection without restarting Creoson server or avoiding the server overload in the first place? I've attempted to "sleep" the code a bit to slow it down as a test and the same error occurs. Seems like the error trigger is time based and not related to the amount of requests being sent. i.e. when the code is running continuously for ~5 minutes or more. 

I'd even take advice like "don't make any more than X calls in Y timeframe" if someone could provide the X and Y variables. Thanks for any help!
-Brian

bperryrm, 2025-08-27:
> You should keep the sessionID returned on connection.  Otherwise - you might try
> putting a catch on this error and executing another connection to get a new
> sessionID. 
> […](#)

I’m able to catch the error as an exception but when I try to create a new creopyson object with new sessionID and connect to it, the “connect” method fails with another error “NewConnectionError” saying the new connection was “refused by the remote host.” I’ll see if I can create a slimmer version of code to debug it more reliably. 

Side note: this tool you’ve created is awesome, has saved me lots of hours of work!

davidhbigelow, 2025-08-27:
> [https://ci3.googleusercontent.com/meips/ADKq_NaFST-AELS_hpcGnkxBEwpJBE-NonJnofVZrYQJwYeLqWWtu1da6DIhVHXm2C760Pjfc7ynAU4Gc8n9p1YKHc2D8jXipqrc629dBxn5ZObp=s0-d-e1-ft#https://avatars.githubusercontent.com/u/124022981?s=20&v=4]bperryrm
> commented.[https://ci3.googleusercontent.com/meips/ADKq_NaeNDd3fQ28mCvUkWXGggQJx9W_LgeWoNz3xmcMTD4AHxlgSBDB_90LvG6yvrziP6t0WSDBLf86NpnpQLYwg9ruceZoM_fUfKB0yqz3PtlNZkAW9PtStUz9mipPrsbHmmONl-CKdQv3DrwtaLJWhNdrbFTqxlTycTG3iWERN9zRUlgRrsROxAvTOzSlBB6v_XLnsi2ZzoI_ewLt8FDr4eNAmABODIJjYmCdmjoODxtxMjVtqYuxZiE=s0-d-e1-ft#https://github.com/notifications/beacon/AAMSRJIGGLUDCJDBOBCRDRD3PYQUPA5CNFSM6AAAAACE7LKTZOWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTWAQNH5A.gif]Message

Try disconnecting with the previous sessionID - and forcing a reconnect on the
try/catch when this occurs.

davidhbigelow, 2025-08-28:
> If I try disconnecting after an exception is already thrown, it just
> throws another exception and can never recover. Then I need to forcibly
> stop and restart the server in the CreosonSetup.exe window.
> I put together a short sample that makes the exception come up at random
> times. Sometimes have to run it 3 or 4 times before getting the failure.
> Would appreciate any pointers on fixing.
> `import creopyson
> nRequestCount = 0
> creo = creopyson.Client()
> creo.connect()
> if creo.is_creo_running():
> 		diActiveModel = creo.file_get_active()
> 		sActive = diActiveModel.get("file")
> 		while nRequestCount < 4000:
> 			print("Request count # {0}".format(str(nRequestCount)))
> 			nRequestCount +=1
> 			bParamExists = creo.parameter_exists("TEST_PARAMETER1", sActive)
> 			if creo.is_creo_running(): #Only here to stress test
> 				nRequestCount +=1
> 			if bParamExists:
> 				arr_params = creo.parameter_list("TEST_PARAMETER1", sActive)
> 				print("Parameter List {0}".format(str(arr_params[0].get("name"))))
> 				arr_params = creo.parameter_list("TEST_PARAMETER2", sActive)
> 			else:
> 				creo.parameter_set("TEST_PARAMETER1", value="TEST_VALUE_123", file_=sActive)
> 				creo.parameter_set("TEST_PARAMETER2", value="TEST_VALUE_123", file_=sActive)
> 		print("No Failure found")
> 	except Exception as err:
> 		#This is where the initial error occurs
> 		print("ERROR1: " + str(err))
> 		creo.disconnect() #*****unhandled exception here*****
> 		creo.connect()
> except Exception as err:
> #This catches the second error
> print('ERROR2: ' + str(err))`

Hmmm.. looks like you are stressing it out?

Have you tried to start the process from exec() and track the PID/TaskID
and forcibly kill and restart it?

davidhbigelow, 2025-08-28:
NOTE: This may be a Creo JLINK / memory problem - we have noticed this in
the past.  Our "fix" was to force a kill/restart of the server (like every
20 sessionID transactions) to be more "safe" about the interface getting
locked up. That has never been a great thing to deal with... takes a couple
of seconds to kill/restart/reconnect.

closed 2025-11-08 by Zepmanbc